ASU-ASCEND/Fall-2022 #3: All components classified and modeled

darthtaterdude, 2022-08-19:


bdweber1, 2022-09-05:
- [ ] Talk to SM Director Ben about what components need to be where
- [ ] Talk to SM Director David about what components need to be where
- [ ] Talk to Subsystem Lead Anyell about what components the PCB will be using

bdweber1, 2022-09-05:
Meteorology:
GPS: https://grabcad.com/library/ardusimple-simplertk2b-heading-1
Humidity: https://www.farnell.com/datasheets/3167872.pdf (Needs to be modeled)
Thermistor: https://www.amphenol-sensors.com/hubfs/Documents/AAS-920-308B-Thermometrics-NTC-TypeBR-100714-web.pdf (Needs to be modeled)
RTD: https://prod-edam.honeywell.com/content/dam/honeywell-edam/sps/siot/en-us/products/sensors/temperature-sensors/rtd-sensors/hel-700-series/documents/sps-siot-hel-700-series-thin-film-platinum-rtd-productsheet-009080-1-en-ciid-150504.pdf (Needs to be modeled)

bdweber1, 2022-09-05:
ADCS:
DC motor: https://grabcad.com/library/dys-sf2207-brushless-motor-1
Micro servo: https://grabcad.com/library/servo-mg92b-1
IMU: https://grabcad.com/library/adafruit-bno055-1

bdweber1, 2022-09-05:
General:
Arduino: https://grabcad.com/library/arduino-nano-every-1

bdweber1, 2022-09-18:
Completed enough for now.

closed 2022-09-18 by bdweber1