ASU-ASCEND/Fall-2022 #32: Structure

bdweber1, 2022-09-05:
Task List:

- [ ] [2U Structure](https://www.isispace.nl/product/2-unit-cubesat-structure/)

GanttStart: 2022-09-05
GanttLab: 2022-09-16

bdweber1, 2022-09-05:
1U for each science mission

JoeyBuch, 2022-09-15:
Regarding CubeSat Kit Structures, where should modeling team go to find necessary dimensions? Stated 1U for each science mission: do we have 2 science missions? Can start, but unsure of constraints

Johnny-Realman, 2022-09-15:
Is there any information from the previous team

JoeyBuch, 2022-09-15:
Going to go off basis of 1U being 100x100 mm floor with 113.5mm height

bdweber1, 2022-09-18:
Science Missions:
- 1U for the Meteorology science mission. Atmospheric profiling and the sensors needed for that.
- 1U for the Attitude Determination and Control System science mission.

Previous structure had no documentation (that's on me). Essentially was just slap a foam box together and move things around until they worked. No model constraints then.

1U basis is correct for 100mm x 100 mm floor and 113.5mm height.

bdweber1, 2022-09-30:
Progress has been made. Will be updated in #17

closed 2022-09-30 by bdweber1